RobTillaart/INA226 #2: Incorrect current value for different shunt

zaffresky, 2021-06-21:
Hi, I have tried using your library with the INA226 module. Usually the modules have the 0.01R shunt but the ones I have got has 0.01R shunt. I modified your demo sketch with the shunt value of 0.01 & max current as 8 Amps but I get incorrect results. With  a bus voltage of 3.3V and a load resistor of 5R the current should be around 0.66A (also verified by the DMM) but the sketch outputs 1.25A. I wonder if it is an issue with the library or do some other variable values also need to be modified when using a different shunt?

RobTillaart, 2021-06-21:
[image]

started test with the above breakout, 

SHUNT = 0.002 ohm
LOAD = resistor 194 ohm

I = V / R = 5.0 / 194 = 0.02577...

```
BUS		SHUNT		CURRENT		POWER
5.088750	0.000047	0.018616	0.000992
5.088750	0.000047	0.018616	0.000877
5.088750	0.000045	0.017548	0.000954
5.088750	0.000047	0.019531	0.000992
5.086250	0.000050	0.019531	0.000992
5.087500	0.000050	0.019531	0.000954
5.088750	0.000047	0.018616	0.000954
5.086250	0.000050	0.019531	0.000954
5.088750	0.000050	0.019531	0.000954
5.087500	0.000047	0.018616	0.000954
5.088750	0.000047	0.018616	0.000954
5.087500	0.000053	0.020447	0.001068
```

DMM measurements
Measured 5V == 5.07  ==> OK
Measured amps = 25.7mA ==> OK

The voltage of the shunt is very low. If we calculate the I from that value we should get

I = 0.000050 / 0.002 = 0.025 = 0.25mA ==> OK

The function **getCurrent()** returns 20% too low which implies the problem is in the calibration.
(note the measurement is on the edge of the sensor, so I might need to change my setup)
 
so the first suspect is the **INA.setMaxCurrentShunt(...);** function.

RobTillaart, 2021-06-21:
- rewrote **setMaxCurrentShunt()**
- added **float getShuntVoltage_mV()** for scaling output
- added **float getPower_mW()** for scaling output
- added **float getCurrent_mA()** for scaling output

```
BUS	SHUNT	CURRENT	POWER
3.321	0.030	15.000	50.000
3.321	0.027	13.800	45.000
3.322	0.030	15.000	45.000
3.322	0.030	15.000	50.000
3.324	0.032	16.300	52.500
3.322	0.032	16.300	45.000

```

Note we are still on the edge of accuracy

3.322 V  / 194 ohm => 17.1 mA  so the measurements are up to ~8%  (too low)

```
BUS	SHUNT	CURRENT	POWER
5.088	0.050	23.800	127.500
5.089	0.052	26.300	122.500
5.090	0.050	25.000	127.500
5.089	0.045	22.500	127.500
5.089	0.050	25.000	127.500
```

5.09 / 194 = 26.2 mA  so the measurements are up to ~9%  (too low)

```
BUS	SHUNT	CURRENT	POWER
6.650	0.067	33.800	225.000
6.650	0.065	32.500	225.000
6.650	0.067	33.800	225.000
6.649	0.067	32.500	225.000
6.649	0.065	32.500	225.000
6.649	0.067	33.800	225.000
6.648	0.065	32.500	217.500
6.648	0.062	31.300	210.000
```

6.65 / 194 = 34.3 mA  so the measurements are up to ~9%  (too low)

```
BUS	SHUNT	CURRENT	POWER
10.047	0.105	52.500	515.000
10.047	0.105	52.500	515.000
10.047	0.105	52.500	527.500
10.046	0.102	51.300	515.000
10.046	0.105	52.500	527.500
10.047	0.105	52.500	527.500
10.049	0.105	52.500	515.000
10.049	0.107	53.800	540.000
```

10.047 / 194 = 51.8 mA  so measurements are better  (probably because we are further from the edgy measurements)

```
BUS	SHUNT	CURRENT	POWER
12.539	0.135	67.500	847.500
12.539	0.132	66.300	847.500
12.541	0.135	67.500	847.500
12.540	0.132	66.300	832.500
12.541	0.135	67.500	832.500
12.544	0.132	66.300	832.500
12.541	0.135	67.500	847.500
```

**WARNING: resistor got HOT**

12.541 / 194 =  64.6 mA  the measurements are ~3% too high

I will make a new branch so you can test the new code.

RobTillaart, 2021-06-21:
@zaffresky 
new brach created - https://github.com/RobTillaart/INA226/tree/develop

please verify if it improves your measurements.

RobTillaart, 2021-06-21:
> Hi, I have tried using your library with the INA226 module. Usually the modules have the 0.001R shunt but the ones I have got has 0.01R shunt. I modified your demo sketch with the shunt value of 0.01 & max current as 8 Amps but I get incorrect results. With a bus voltage of 3.3V and a load resistor of 5R the current should be around 0.66A (also verified by the DMM) but the sketch outputs 1.25A. I wonder if it is an issue with the library or do some other variable values also need to be modified when using a different shunt?

The line that you should modify is 
```cpp
  INA.setMaxCurrentShunt(8.0,  0.01);
```
- expect max 8 Amp (order of magnitude rounded up)
- shunt == 0.01 ohm

zaffresky, 2021-06-21:
I tried the test branch but still the current values I get make no sense. E.g. with 5V bus voltage I get: 

```
4.990	1.130	113.000	575.000
4.987	1.133	113.000	575.000
4.986	1.133	113.000	575.000
4.985	1.137	114.000	575.000
```

This is with the modification you just mentioned 
`  INA.setMaxCurrentShunt(8.0,  0.01);`

zaffresky, 2021-06-21:
The module I am using looks like this

[image]

zaffresky, 2021-06-21:
With 10 ohm load and 5V bus voltage I get

4.777	10.452	1045.000	5000.000
4.776	10.452	1045.000	5000.000
4.776	10.450	1045.000	5000.000
4.774	10.447	1045.000	5000.000
4.775	10.447	1045.000	5000.000

Current acc to DMM = 451mA

seems like the sensor output is almost double the actual current

zaffresky, 2021-06-21:
A question. Why do we need to specify the max current  we expect in the `INA.setMaxCurrentShunt(8.0, 0.01);` function? Does it has influence on the current measurement results?

RobTillaart, 2021-06-21:
Bus voltage = 4.7
R = 10 ohm so current should be 0.47 Ampere ~ 451mA so your DMM is OK.

451 mA * 0.01 R = 4.51 volt expected over the shunt, 
the code shows 10.45  **so the shunt voltage is not correct.**

RobTillaart, 2021-06-21:
> A question. Why do we need to specify the max current we expect in the `INA.setMaxCurrentShunt(8.0, 0.01);` function? Does it has influence on the current measurement results?

It is needed to calculate the value for the calibration register. Technically it allows to get more significant digits.

zaffresky, 2021-06-21:
> Bus voltage = 4.7
> R = 10 ohm so current should be 0.47 Ampere ~ 451mA so your DMM is OK.
> 
> 451 mA * 0.01 R = 4.51 volt expected over the shunt,
> the code shows 10.45 **so the shunt voltage is not correct.**

So does that mean the module is faulty or shunt resistor isn't a 0.01 ohm?

RobTillaart, 2021-06-21:
Yes, 
The sensor seems to read the right voltage for the bus so I expect that the ADC does work correctly. 
So my prime suspect is the shunt resistor.

I just pushed a new version of the lib in develop stream, including an example to read the values in the registers.
That might help to understand what is happening

zaffresky, 2021-06-21:
After uploading the new dump register sketch the serial output remains blank. I have checked the connections and everything is fine. If I reload the INA_demo sketch the serial output works fine :(

RobTillaart, 2021-06-21:
Assuming the current is indeed 451 mA and the voltage is 10.45 mV we can calculate the resistor value

R = 10.45 / 451 = 0.0232 ohm   ==> could be 0.02  or  0.022 

What happens if you set in the demo sketch a value of 0.02 for the shunt  (even if it has a print 0.01)
```cpp
INA.setMaxCurrentShunt(8.0, 0.02);
```

RobTillaart, 2021-06-21:
> After uploading the new dump register sketch the serial output remains blank. I have checked the connections and everything is fine. If I reload the INA_demo sketch the serial output works fine. Strange behaviour :(

What board do you use? ESP?

zaffresky, 2021-06-21:
I am using Pro Micro 5V. 

With the shunt value of 0.02 the current output is more realistic:

```
4.782	10.465	523.000	2500.000
4.780	10.465	523.000	2500.000
4.779	10.457	523.000	2500.000
4.779	10.455	523.000	2500.000
4.777	10.452	523.000	2500.000
4.777	10.450	523.000	2500.000
4.775	10.447	522.000	2500.000

```

The DMM reads 449mA so still a difference of ~ 70 mA.

zaffresky, 2021-06-21:
Here is the output with 3V & same 10 ohm resistor

```
3.051	6.675	334.000	1025.000
3.052	6.672	334.000	1025.000
3.052	6.672	334.000	1025.000
3.054	6.675	334.000	1025.000
3.054	6.683	334.000	1025.000
```

The DMM reads 290 mA. So difference is smaller this time

RobTillaart, 2021-06-21:
And if you just adjust it to make it match - that is what calibration is in the end ...

```
INA.setMaxCurrentShunt(8.0, 0.0232);
```

RobTillaart, 2021-06-21:
I have changed the dump_register.ino example so it dumps the register in loop()

Should generate some output now...

RobTillaart, 2021-06-21:
> Here is the output with 3V & same 10 ohm resistor
> 
> ```
> 3.051	6.675	334.000	1025.000
> 3.052	6.672	334.000	1025.000
> 3.052	6.672	334.000	1025.000
> 3.054	6.675	334.000	1025.000
> 3.054	6.683	334.000	1025.000
> ```
> 
> The DMM reads 290 mA. So difference is smaller this time

The current scales with the voltage so the error scales too

zaffresky, 2021-06-21:
Yes the serial output works now :)

for 3V & 10 ohm resistor, DMM output 289 mA

Serial output:

```
	REG	VALUE	VALUE_X
	0	16679	4127
	1	2689	A81
	2	2437	985
	3	41	29
	4	339	153
	5	256	100

```

zaffresky, 2021-06-21:
> The current scales with the voltage so the error scales too

Makes sense :)

RobTillaart, 2021-06-21:
From the datasheet

[image]

config register 0 == OK
shunt register 2689   x 2.5uV = 6.7225 mV  ( seen above == OK )
bus voltage register 2437  x 1.25 mV =3.046 V  (roughly seen above == OK)
power register  41 * 25 mW = 1.025 Watt (seen above OK)

current register 339  x   LSB (??)  = 334 ??   so LSB is ~ 1mA
calibration register 256 =  0.00512 / (LSB * Rshunt)   = 0.00512 / (0.001 * 0.02)   ==> shunt == 0.02 

Sofar I understand the INA226, all values make sense except the Rshunt

RobTillaart, 2021-06-21:
So after several hours my conclusion is that I expect the library to be OK and that the value of the shunt on your sensor is not labeled right. 
Action: order some heavy duty resistors to test a wider range of voltages and currents

zaffresky, 2021-06-21:
I am thankful for your time and support. I also believe shunt is the issue here. I have basically two of these modules and both of them have the same issue. 

I have a few 50W power resistors (1 ohm, 5 ohm and 10 ohm) so would that work? What test plan do you suggest?

RobTillaart, 2021-06-22:
Sorry, the action I mentioned was for me to take.

Testing should at least mimic the way you intend 
to use the sensor. As library writer the edge cases are also important. E.g. I got some small negative values when no load was connected. Noise of the ADC probably, the lib should be able to handle this properly.

RobTillaart, 2021-06-22:
> I have basically two of these modules and both of them have the same issue.

Could be same production run.

> I have a few 50W power resistors (1 ohm, 5 ohm and 10 ohm) so would that work? What test plan do you suggest?

Keep in mind the max current and voltage the system can handle . (check datasheet for specs)

RobTillaart, 2021-06-22:
0.1.1 released 

If the issue is solved for you, you may close the issue.

zaffresky, 2021-06-22:
> Sorry, the action I mentioned was for me to take.
>
> Testing should at least mimic the way you intend to use the sensor.
>
> As library writer the edge cases are also important. E.g. I got some small [...]

Yes  I also encountered that in my testing

zaffresky, 2021-06-22:
> 0.1.1 released
> 
> If the issue is solved for you, you may close the issue.

I suppose with the existing shunt resistors, I won't be able to get much better results.

closed 2021-06-22 by zaffresky