RobTillaart/INA226 #7: float INA226::getCurrent() does not handle negative values properly

RobTillaart, 2021-08-07:
@FrankTmmrmn

See begin of discussion in Pull Request #6 

Essence of the problem

> in my application - NiMh battery analyzer the current is positive when a load is connected but i don't get a negative number when I am charging.

RobTillaart, 2021-08-07:
update: development branch created

RobTillaart, 2021-08-07:
@FrankTmmrmn

- Pushed a fix to the development branch. Can you please verify if the values are shown correctly in your application?
- added a line with units to **INA226_demo_2.ino** to make them explicit in the Serial Output.
- commented the "ID lines in setup()" so the output of Serial Plotter is improved. See below.
  note the POWER and POWER2 line are pretty close (but that is only a positive current test).
  BusVoltage changed between ~5.0 V and ~0.5V and all reacted as should.

[image]

RobTillaart, 2021-08-07:
Notice that the delta has some relative larger bumps 
The cause is that POWER uses less bits than the POWER2 calculation. 
For small values this gives a relative larger DELTA. 

Might just need some more averaging...

RobTillaart, 2021-08-08:
@FrankTmmrmn
Can you verify if the **develop** branch works?

FrankTmmrmn, 2021-08-09:
Hi Rob
I have compiled and loaded the code INA226_demo_2
here is the output of the serial monitor

POWER2 = busVoltage x current
BUS CURRENT POWER POWER2 DELTA
26.692 761.300 659.20 20321.00 19661.80
26.681 760.000 684.20 20277.75 19593.55
26.674 762.500 676.70 20338.74 19662.04
26.695 760.000 626.70 20288.20 19661.50
26.660 762.500 569.20 20328.25 19759.05
26.656 760.000 599.20 20258.75 19659.55
26.651 761.300 631.70 20289.60 19657.90
26.647 758.800 594.20 20220.12 19625.92
26.644 760.000 589.20 20249.25 19660.05
26.666 760.000 606.70 20266.35 19659.65
26.747 571.300 2175.30 15280.85 13105.55
switch to charging
27.066 6494.900 1590.00 175792.57 174202.57
27.102 6496.100 1557.50 176060.54 174503.04
27.087 6493.600 1625.00 175895.39 174270.39
27.090 6494.900 1590.00 175946.84 174356.84
27.096 6493.600 1625.00 175952.20 174327.20
27.125 6496.100 1560.00 176206.71 174646.71
27.102 6493.600 1625.00 175992.79 174367.79

kind regards

RobTillaart, 2021-08-09:
mmm does not look like solved. 

1) Are you sure you have downloaded the **develop** branch? 
https://github.com/RobTillaart/INA226/tree/develop

I notice the header of the **INA226_demo_2.ino** sketch looks not updated as I added mW and mA etc.
This could indicate you 

2) Do you have multimeter values for current, to see what the actual values are?
That would allow me to do some reverse reasoning that might help finding the cause.

3) Do you have a sketch of your hardware setup, so I might be able to recreate it (when time permits)

FrankTmmrmn, 2021-08-09:
Rob
cpp has the contents of the header file
frank

FrankTmmrmn, 2021-08-09:
sorry my mistake its only in the comment section that you mention the .h
file twice
seems to be working fine now Rob.

this is the output of the serial monitor

startup in charging mode

POWER2 = busVoltage x current
 V mA mW mW mW
BUS CURRENT POWER POWER2 DELTA
27.081 -493.700 270.30 -13370.01 -13640.31
27.081 -493.700 270.30 -13370.01 -13640.31
27.081 -493.700 272.80 -13370.01 -13642.81
27.081 -493.700 275.30 -13370.01 -13645.31
27.061 -432.500 5151.40 -11703.99 -16855.39

switch to load 1
26.712 570.000 2122.80 15226.12 13103.33
26.710 570.000 2115.30 15224.70 13109.40
26.706 570.000 2112.80 15222.56 13109.76
26.705 570.000 2115.30 15221.85 13106.55
26.702 570.000 2110.30 15220.42 13110.12

switch to load 2
26.342 751.300 144.20 19791.12 19646.92
26.341 751.300 139.20 19790.18 19650.98
26.341 751.300 139.20 19790.18 19650.98
26.341 751.300 129.20 19790.18 19660.98
26.341 751.300 131.70 19790.18 19658.48
26.341 751.300 139.20 19790.18 19650.98

switch to load 3
26.082 1113.800 2845.60 29050.69 26205.09
26.082 1115.000 2858.10 29081.99 26223.89
26.082 1113.800 2850.60 29050.69 26200.09
26.082 1113.800 2843.10 29050.69 26207.59
26.081 1113.800 2840.60 29049.29 26208.69
26.081 1115.000 2853.10 29080.59 26227.50

switch to charging mode

26.870 -496.200 222.80 -13332.89 -13555.69
26.870 -496.200 232.80 -13332.89 -13565.69
26.870 -496.200 225.30 -13332.89 -13558.19
26.870 -496.200 225.30 -13332.89 -13558.19
26.870 -496.200 225.30 -13332.89 -13558.19
26.870 -496.200 237.80 -13332.89 -13570.69

RobTillaart, 2021-08-09:
It looks quite a bit better as the current is charging / draining dependent.
I will merge it into the master as it is an improvement for sure.

In my tests the POWER and POWER2 were always (~) equal in size. 
Any idea why it isn't in your test runs?

RobTillaart, 2021-08-09:
Merged into master. 
If there are no more questions / problems you can close this issue.

FrankTmmrmn, 2021-08-09:
No idea at the moment Rob
I didn't check any of the returned values with a calibrated meter, I will
have to dig deeper in the different commands and the setup of the INA in
general. First step was to get values out of the sensor and display them on
the LCD and store everything on a SD card.
I will look at it later, if i find something i will let you know
gr
Frank

RobTillaart, 2021-08-10:
OK, thanks again for pointing out the bug. 
For now I will close this issue and just create a new one if needed, or reopen this if it seems related.

closed 2021-08-10 by RobTillaart