RobTillaart/INA226 #13: Should power reading be signed?

morrowwm, 2022-02-03:
The current register is a signed (two's complement) 16 bit value.
Could getCurrent() and getShuntVoltage() simply use signed integers to read the registers?
```
float INA226::getCurrent()
{
  **int16_t** val = _readRegister(INA226_CURRENT); // current register is signed (two's complement) 16 bit
  return val * _current_LSB;
#endif
```
Strangely, it looks like the power register is unsigned, even though current is signed and equation 4 in the datasheet says:
[image]

RobTillaart, 2022-02-05:
Power is current x voltage.  If the voltage is reversed the current is reversed, and both - signs will cancel out.

Yes that should be simpler, but I need a test setup with a negative load (e.g. battery loader) to verify it.

morrowwm, 2022-02-05:
Measuring power in to and out from a battery is my application. I can confirm power is always positive. I simulated negative current flow by just switching the leads across the shunt resistor. Both Vshunt and current are negative, but power is positive.
I can deal with my requirements outside the library. Maybe record net amp-hours instead of power.

RobTillaart, 2022-02-05:
I tested in a small sketch some values and the math looks OK, so I fixed it in the current develop branch.

commit d3e44ce8 (2022-02-05): fix #13 sign handling
commit 9638e6db (2022-02-08): fix #11 normalize routine (#12)

* fix #11 normalize routine
* fix #13 sign handling
* add releaseNotes.md
closed 2022-02-08 by RobTillaart