RobTillaart/INA226 #16: Higher current shunt issue

Juicy-AUS, 2022-07-03:
I’m using an external 40A 75mV shunt with the ina226. When I input the values to the `ina.setMaxCurrentShunt(40.0, 0.001875)` I’m unable to measure any current. 
Interestingly, if I use 20a I am able to read small currents (200-400mA) fairly accurately. I’ve yet to setup a test to pull larger currents. 
Is having such a large max current reducing the A/mV too much that it’s unable to register any current or something else?

RobTillaart, 2022-07-03:
Thanks for reporting,

As I am quite busy with some other libraries it might take some days before I can investigate.
Think I need to reread the datasheet carefully for this one. 

- which platform do you use?  UNO / ESP?
- can you post a schema of your setup?
-

Juicy-AUS, 2022-07-03:
Thank you. 
Im using an esp32. 
I will try to put a drawn schema together. 
I’m using the same breakout board your were using for testing in issue #2 but when I first didn’t get any readings, I decided to remove the surface mounted R002 resistor from the board, thinking since both resistances were in parallel, it may cause issues. 
I will try to get together a drawn schema for you. I’m using the same breakout module you were testing with in issue #2. Vin connected to one side of shunt. Vout connected to other side. Using a switch mode power supply positive connected with Vin at shunt and an led strip connected to with vout at shunt which then goes to ground of power supply. 
Voltage readings are fairly decent (seems to be out +/- 350mV but that could also be my dmm as I need new leads) and as mentioned earlier, fairly accurate current readings at <1A with 20A setting but feel that might start to swing away with higher currents.

RobTillaart, 2022-07-03:
ina.setMaxCurrentShunt(40.0, 0.001875)

I had a quick look in the code and it is limited to 20Amps. The call should return false because params are out of range. You could test that.

Note for myself: check documentation.

Juicy-AUS, 2022-07-03:
I was worried about that. Call seemed to return just 0.00A. 
Is the limit changeable?

RobTillaart, 2022-07-03:
You are free to change the code as it is open source, I do not know without further investigation if and how well that will work.
Remember that the sensor has a physical limitation. _Advice read the datasheet, twice_

Keep me informed of your insights, might help to improve the library and documentation.

Juicy-AUS, 2022-07-03:
After posting I realised I can just go to the .cpp and change so I will try that and report back. 
I looked through the data sheet and didn’t see anything about a current limit. I suppose it’s not really measuring current, just the voltage drop. There is a max voltage between vin and vout of 81mV so if shunt resistance * max current was above that, I suspect chip might let the smoke out. But mine should only be 75mV at 40A

RobTillaart, 2022-07-05:
you may uncomment line 194  **#define printdebug true** to get some numbers.

Juicy-AUS, 2022-07-10:
I changed line #198 to max current > 40 and all has worked fine. I’m yet to confirm accuracy of current but will get to that.

RobTillaart, 2022-07-11:
mmm, the limit of 20 A was because the sensor can handle 20 A.
Code calculates the accuracy - LSB - from this value.

There might be no reason to prevent the math if current is "too high", just return a warning if the value is above 20 A.
This way the user can see if and where setting the internals failed.

Must dive into the datasheet and code to recall the details.

Just to hold my "current" thoughts, I'm thinking of this.
```cpp
int INA226::setMaxCurrentShunt(float maxCurrent, float shunt, bool normalize)
{
  // #define printdebug true
  uint32_t calib = 0;
  uint32_t factor = 1;

  if (maxCurrent < 0.001) return INA_ERR_MAXCURRENT_LOW;
  if (shunt < 0.001) return INA_ERR_SHUNT_LOW;

  // do al the math

  if (maxCurrent > 20) return INA_ERR_MAXCURRENT_HIGH;
  return INA_ERR_NONE;
}
```

RobTillaart, 2022-08-24:
@Juicy-AUS 
Sorry it took so long to respond,

>  But I think the breakout board, due to size of tracks, shunt resistor type etc is limited though. 

That is definitely true

I'm going to create a develop branch without the limit check, will set a warning or so,  and update the documentation to reflect these new insights.

RobTillaart, 2022-08-26:
@Juicy-AUS 
Created a develop branch for 0.4.0 which includes a 80 mV shuntVoltage check instead of a maxCurrent check
- https://github.com/RobTillaart/INA226/tree/develop

If you have time could you check if it works for you?

closed 2022-08-28 by RobTillaart

RobTillaart, 2022-08-28:
@Juicy-AUS 
This new version should work for you, otherwise reopen the issue.
If there are other problems, please open a new issue.

Juicy-AUS, 2022-08-28:
Thank you @RobTillaart. I’ve been a little busy but will put in this version and test when I can. I appreciate your work sir.

commit 7e188ca6 (2022-08-28): fix #16 - improve calibration and working range (#19)

* fix #16 - improve calibration and working range
* add example + documentation about normalization flag