RobTillaart/INA226 #20: Overflow current >= 32768  (int16)

felipengeletrica, 2022-08-26:
When trying to do with currents above 33768 mA. 
This is probably caused by using 16-bit variables and the maximum current limit.

Thank you in advance for your time and for this library.

RobTillaart, 2022-08-27:
Thanks for this issue,
Think your analysis is on a right track.

RobTillaart, 2022-08-27:
@felipengeletrica 

Did a small test and it seems to happen when you 'normalize' the current.

The maxCurrent is calculated from the 'normailized' LSB which ends up being 32768 * LSB = 32768 mA 
In the code a normalized LSB is always  1 (with a power of 10)

Run this sketch shows the difference:

```cpp
//
//    FILE: INA226_test.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: test sketch
//     URL: https://github.com/RobTillaart/INA226

#include "INA226.h"
#include "Wire.h"

INA226 INA(0x40);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  Wire.begin();
  if (!INA.begin() )
  {
    Serial.println("could not connect. Fix and Reboot");
  }
  INA.setMaxCurrentShunt(79, 0.001, true);
  Serial.println(INA.getMaxCurrent(), 3);
  INA.setMaxCurrentShunt(79, 0.001, false);
  Serial.println(INA.getMaxCurrent(), 3);
}

void loop()
{
}
```

Output:
32.768
79.000

Please check if this solves the issue.

RobTillaart, 2022-08-27:
I will add the above sketch to next release to show the diff between normalize and not.

closed 2022-08-28 by RobTillaart

RobTillaart, 2022-08-28:
@felipengeletrica 
Closed the issue as normalization explained the behavior - see updated documentation.

felipengeletrica, 2022-08-28:
Working now! Very thanks! 🚀

RobTillaart, 2022-08-28:
You're welcome!