RobTillaart/INA226 #21: Voltage and current reading wrong by large margin

Chick92, 2022-09-04:
Hi, 

I suspect my issue is down to a diffy INA226 board / IC

I have the module wired in to a 16.6V battery, with load around 5W (can't give you an accurate reading on what it is without dismantling the wiring harness. 

Voltage from the INA226 reads 11.165 V and current reads on the order of 2 A. 

My multimeter shows the true voltage of 16.6V, and it's been powered up for an hour or so now with minimal change in measured voltage from the multimeter, so i know it's not drawing anything close to 2 A.

The voltage from the INA226 was reading 10.2 a while ago. 

I have a 0.001 ohm shunt resistor, but my Fluke MM can't measure the resistance accurately to know how good this is. 

Any ideas? It's connected up following this wiring example - https://wolles-elektronikkiste.de/en/ina226-current-and-power-sensor

This is my output - 

BUS	SHUNT	        CURRENT	POWER

11.165	0.160000	2682.400	43600.000
11.165	0.165000	2682.400	43600.000
11.165	0.165000	2682.400	43600.000
11.165	0.165000	2682.400	43600.000
11.165	0.165000	2682.400	43600.000
11.165	0.240000	2682.400	43600.000
11.165	0.165000	2682.400	43600.000
11.165	0.160000	2682.400	43600.000
11.165	0.160000	2682.400	43600.000
11.165	0.240000	2682.400	43539.996

Cheers!

RobTillaart, 2022-09-04:
Thanks for the issue, might take a few days before I can investigate.

RobTillaart, 2022-09-04:
The link in your post points to another library.
Do you get the same results using that library?

RobTillaart, 2022-09-04:
Which sketch do you use?
Have you adjusted the **setMaxCurrentShunt()**

Chick92, 2022-09-04:
Hey Rob, 

Yeah similar results, although not the same when using that library. I used that link for illustrative purposes as that's how it's wired up. 

That data is from the ina226_demo.ino file, and changing the values in setMaxCurrentShunt() has no effect. I suspect there's a problem with the module...

RobTillaart, 2022-09-05:
### Analysis

Your power = 16.6 V
INA measures = 11.165 V 
==> drop of ~5.5V that's one third

load should be 5W so that is about 300mA, 
Measured 2A 

Possible causes:
- the power supply cannot handle the load?
- bad wiring?
- the INA226 board fails

### test 0
- check if wires are connected well. Use a fresh breadboard if needed.
- check the soldering on the INA226 board. 

### test 1
Do you have a DMM to measure the current and voltage when you connect the load directly to the 16.6 V?
- does the voltage drop too?
- what is the measured current?

### test 2
Use a 5V power supply and use as load a resistor of 1 KΩ
INA should report 5V and 5 mA.
Do you see those numbers?

Chick92, 2022-09-05:
The battery is rated for 100C and is 4S 18 Ah. The INA226 is on the input between the battery and load, all voltages are what i'd expect them to be when measured with a multimeter on the INA226 and at multiple points down stream. 

No voltage drop, and not able to measure current without dismantling the setup. 

Test 2 seems like the most sensible one, as i'm going to replace the 226 with a 219 (although on a downstream part, as it won't be able to measure the full load).

Accurate current and power would be nice, but voltage is all i really need as this is just to determine when the robot is running low and needs to be recharged. 

Ben

RobTillaart, 2022-09-05:
> ... voltage is all i really need as this is just to determine when the robot is running low and needs to be recharged.

Did you consider a voltage divider e.g. 1 in 4 so the 16.6V => 4.15V  == ~850 Arduino ADC
When the voltage drops e.g 10% to 15 V => 3.75 V == ~765 ADC
Even a drop of 2% is detectable 16.6V => 16.3V => 4.08V == ~835 ADC

RobTillaart, 2022-09-05:
> The INA226 is on the input between the battery and load, all voltages are what i'd expect them to be when measured with a multimeter on the INA226 and at multiple points down stream.

That makes the INA226 prime suspect. Test with INA219 is a good idea.

For current measurement an ACS712 works pretty well, different ranges 5, 20, 30A available.
- https://github.com/RobTillaart/ACS712

Chick92, 2022-09-05:
Yeah i'll probably go with that. I've found the INA226 to be notoriously unreliable (this is the second of three i've tried, different design too). I also have a product that uses INA219 modules and had to rebuild 6 or 7 of the PDB units to replace the INA219's as none of them were working! 

Shame really because when they do work they're a neat solution for both voltage and current.

closed 2022-09-05 by Chick92

RobTillaart, 2022-09-05:
OK, success!

RobTillaart, 2022-09-06:
@Chick92 

What voltage do you power the INA226?
A thought came in my mind: 16.6V / 5V * 3.3V = 10.956V  is almost your 11.156 (within 2%).
Need to check the datasheet about VCC