RobTillaart/INA226 #23: Ina226 current read 0.00 random time

poorhackerman, 2023-03-15:
Hi!

Quick question, any Idea why after some time, current readings are 0.00 (when real current is for sure not 0) but in same time voltage is properly measured? Issue appears in several machines after several days of continosus work. Switching arduino off and on fix issue for several days. Thanks!

RobTillaart, 2023-03-15:
No clue at all, not heard before or encountered.
It sounds a bit like a memory leak?

As I do not know the (complexity of) the code it might have other causes 
- heap and stack touching
- dynamic memory ?

First thing i would do is add a heartbeat led to see that the main loop is executing properly even after zeroing of the current.

Second logging the free memory can confirm memory leKs.

Do you use dynamic memory? Malloc/free etc

RobTillaart, 2023-03-18:
Any new insights?

closed 2023-03-29 by RobTillaart

adarpis, 2026-01-16:
> [@adarpis](https://github.com/adarpis) Thanks for reporting back, I will reopen the issue.
> 
> Since the original post I do not recall any similar problems reported. Not for the INA226 nor any of the other INAxxx devices I have libraries for.
> 
> These functions do work:
> 
>     * float getBusVoltage();       //  Volt
> 
>     * float getShuntVoltage();     //  Volt
> 
> 
> And these return 0.00 after several days.
> 
>     * float getCurrent();          //  Ampere
> 
>     * float getPower();            //  Watt
> 
> 
> If you want me to analyze the problem, I need to know a lot more, so can you tell more about the system? what Arduino is used? power supply? what is measured? range? frequency? Version of the library used? Does the problem occur after a fixed time (e.g. exactly after H hours and M minutes ) after restart? Are there more sensors on the Arduino?

@RobTillaart 

I'm using a esp32 customized board, and power supply is switched power based on MP2181 IC. Measuring power of a resistive-inductive load. Version: 0.6.6. Aleatory time. Yes, There are  SPI sensors in another pins.

adarpis, 2026-01-16:
> [@adarpis](https://github.com/adarpis)
> 
> As the getCurrent() and getPower() are code wise pretty identical, I focus first on the getCurrent()>
> ### Code review of getCurrent() function
> 
> ```
> float INA226::getCurrent()
> {
>   int16_t val = _readRegister(INA226_CURRENT);
>   return val * _current_LSB - _current_zero_offset;
> }
> ```
> 
> There are at least three scenario's this function can return zero
> 
>     1. the _readRegister returns zero + _current_zero_offset == 0
> 
>     2. the _current_LSB  is zero + _current_zero_offset == 0
> 
>     3. val * _current_LSB  equals _current_zero_offset .
> 
> 
> ### add 1.
> 
> readRegister can fail to communicate with the sensor. The library provides a simple error code. In such case the function returns zero ==> so getCurrent() becomes zero. Can you add a logging to check if readRegister errors occur?
> 
> ```
> amps = ina.getCurrent();
> int errorCode =ina.getLastError();
> if (errorCode != INA226_ERR_NONE)
> {
>   Serial.print(millis());
>   Serial.print(" ");
>   Serial.println(errorCode);
> }
> ```
> 
> ### add 2.
> 
> Do you modify the currentLSB value runtime?
> 
>     * call to configure()
> 
>     * call to setMaxCurrentShunt()
> 
>     * call to reset()
> 
> 
> ### add 3.
> 
> Do you use the _current_zero_offset ? Can happen just when there is no current (Or correctly offset the noise level). imho this is the less likely scenario.

1. There is not error reading register
2. No, I don't
3. No, I don't

RobTillaart, 2026-01-16:
@adarpis 

A workaround (not solving the cause) could be a build-in retry you could add to the library.

You would need to replace the **getCurrent() in the INA226.cpp file with this version

```cpp
float INA226::getCurrent()
{
  int retries = 3;
  int16_t val = 0;
  while ((retries > 0) && (val == 0))
  {
    val = _readRegister(INA226_CURRENT);
    retries--;
  }
  return val * _current_LSB - _current_zero_offset;
}
```

RobTillaart, 2026-01-16:
@adarpis 

But the configuration is done normally only once - as you do not recalibrate as you stated above?

> I guess for some reason calibration register going to zero..... 

I do not understand why the code snippet is the solution. 
It should at least print something just before the break.
If begin() fails in every call nothing is printed at all.

RobTillaart, 2026-01-16:
> Are there motors or electromagnets involved? Yes, there are motors, but my processor stays OK.

As the problem only occurs after days (randomly) how to be sure there is no electro magnetic interference from those motors?

> I guess for some reason calibration register going to zero.

If you uncomment line 214  **//  #define printdebug true** you get extra output on Serial .

This shows the results of the **INA.setMaxCurrentShunt(27.0F, 0.00281F, true);** call in detail

adarpis, 2026-01-18:
@RobTillaart 
Solution is reset and re-calibration when current and power going to 0.000 and load is plugged. Like a patch. I don't why calibration register reset my MCU are OK, but INA unpredictable reset registers. 3 devices same behavior.

RobTillaart, 2026-01-19:
oops, wrong document, wrong lib.

INA226, sept 2024 datasheet

[image]

similar register layout,
Code that writes to this register guards for out of range and cleans all relevant bits before writing a new value.
So review did not reveal any bug (in two libraries ;)