RobTillaart/INA226 #25: [Question] Which lib is most suitable for a given project?

N0ury, 2023-04-20:
Hi, it's me again...
I'm currently using (for an other project) jarzebski's INA226 lib.
Is it possible to know what are the main differences between this one and jarzebski's?
I can already say I have met some problems with the later on an ESP32-S2 in readRegister16 method
A Wire.beginTransmission and Wire.endTransmission were used for a Wire.requestFrom.
This is unnecessary  and the Soc was hanging.
There was also an unnecessary delay(1) which caused a delay making me loose current variations.

Apart from these two points, what would be the main differences?
Can one be replaced easily by the other?

RobTillaart, 2023-04-20:
Sorry, i am not familiar with jarzebski's lib, have seen it long ago. I would need time to investigate/ compare it, 
So I can't answer those questions.

RobTillaart, 2023-04-20:
Had a very quick look and noticed four things:
- his latest contribution is 9 years ago, so I do not know if it is still maintained 
- 45 forks, which might include improved versions
- the delay(1) you mention is not needed as far as I know, however it could be part of supporting pre 1.0 Arduino versions?
- different license

closed 2023-04-24 by N0ury