RobTillaart/INA226 #26: external high current shunt

jayjar291, 2023-05-03:
I am looking to use a 300A 75mV external shunt I plan to remove the onboard shunt resistor would I just use `setMaxCurrentShunt(300.0, 0.00025)` to calibrate the current readings?

This is how it would be wired
[image]

RobTillaart, 2023-05-03:
Thanks for the question
Think that should work, never done it.
Dont know if the library has some protection build in from my head. Might take a few days to dive into it.

Be sure to use thick wires!

RobTillaart, 2023-05-04:
just a thought that popped up:
after soldering you might want to measure the actual resistance of the shunt.

jayjar291, 2023-05-07:
I looks like it is working. I did have to modify line 200 in INA226.cpp to set the lower resistance.
https://github.com/RobTillaart/INA226/blob/460cbd71d603270f10f4348b596436a2dfe62a00/INA226.cpp#L200

most of the readings I get are correct but it looks like about 2 out of 10 are just random.

RobTillaart, 2023-05-07:
Sounds promising!
Can you describe the conditions when it is failing?
Very high ir liw currents? (Ends of scale)

Did you do some calibration?
How?

What is the resolution of the ADC? 
If it is 10 bits, you get a step size of 300mA in theory, in practice with some noise this can be 2-3 bits == 1000 mA, especially at "low" currents.

Given your results I consider changing that line in a next release. At least add some comments in the readme.md as for mose users it is a safety check.
Another option might be to create a derived class without these tests..need to think it over.
(Same for INA210 ?)

RobTillaart, 2023-05-07:
I make the minimal shunt configurable in the next release referencing this issue.
Furthermore add a note in the readme.md.

```cpp
//  See issue #26
#ifndef INA226_MINIMAL_SHUNT
#define INA226_MINIMAL_SHUNT             (0.001)
#endif
```

RobTillaart, 2023-05-07:
Created PR for this - https://github.com/RobTillaart/INA226/pull/27

RobTillaart, 2023-05-09:
@jayjar291 
Any new insights I can include in the next version?

RobTillaart, 2023-05-22:
@jayjar291 
As no new insights appeared I'm going to close this issue and merge the PR.
Feel free to reopen or add comments if needed.

closed 2023-05-24 by RobTillaart