RobTillaart/INA226 #30: Bus voltage error grows linearly up to 0.5V when measuring 35V

zalexzperez, 2023-12-03:
Just got an INA226 board from China.
Used the demo plotter file and edited it to only display bus voltage measurements. 

Riden RD6006P precise power supply to apply voltages from 0 to 35V at the Vbus pin. UNI-T multimeter confirms the accuracy of the power supply (down to 0.01V at all applied voltages).

Here's a table with the measurements: 
| Set (V) | Max measurement (V) |
|---------|---------------------|
| 0       | 0.03                |
| 0.5     | 0.51                |
| 1       | 1.02                |
| 1.5     | 1.53                |
| 2       | 2.03                |
| 3       | 3.05                |
| 4       | 4.06                |
| 5       | 5.08                |
| 10      | 10.15               |
| 15      | 15.23               |
| 20      | 20.3                |
| 25      | 25.38               |
| 30      | 30.46               |
| 35      | 35.53               |

What causes this offset error? Is there a way to prevent it?
Thanks

RobTillaart, 2023-12-03:
Thanks for the issue, will try to handle it this week.

RobTillaart, 2023-12-03:
(got some thoughts, so a quick share)

Dont know what causes the offset, the only explanation I can come up with is a (systematic) deviation of the shunt. 

Can you do a measurement without anything connected? So no voltage at all? It could indicate some noise.

Can you print the measurements with 4 decimals, especially to get see if the error is linear over the range or not. 

--------

Thinking out loud to prevent it?
Assuming it is an error in the shunt, you could try to change the setup of 1Amp 0.002 ohm
to 1 Amp 0.00197 ohm in the sketch.
Can you give that a try?
Let me know the results.

Another option could be to add correction factors in the library to be able to adjust the precision of the library. For that we need to know if the error is (almost) linear. 
If the error is not linear, it is always possible to use multimap (another library by me) to correct the values (slower as it adds an extra layer of math).

RobTillaart, 2023-12-04:
Given that the other library gives same/ similar error is a sign that the (systematic) error is in the hardware. 

As the sensor makes measurements it needs to have a reference voltage. This is often based upon the voltage the sensor gets to power up.

What is the voltage the sensor gets? 5.000V ?
What happens to the measurements if you vary that voltage? E.g make it 4.900V?

RobTillaart, 2023-12-04:
> I just tried powering it from a separate power supply using 3.3V.
> When measuring 30V, sensor reports 30.436V at 3.3V. When decreasing Vcc all the way to 1.8V, sensor reports 30.426V. That made hardly a difference. Lower voltages than 1.8V shut down the device.

OK, that indicates the sensor uses an internal reference, so we can exclude that as source of the deviation.

>  The shunt resistor is quite close to the rated value. I measured 2.9mV (multimeter) across it while 29.2mA were flowing, so R=2.9/29.2=0.0993Ohms, almost 0.1Ohms.

Note that 0,0993 Ohm  and 0.1 Ohm differ a vert little in absolute terms, but relative it is still about 0.7%
(which equals only half of the 1.5% error)

RobTillaart, 2023-12-04:
[image]

Looking at the diagram in the datasheet (Texas Instruments august 2015) it is not.

> Just got an INA226 board from China.

Don't know if your China board uses a genuine TI sensor or some replica with other specs. 
I have seen other non-genuine sensors that just failed to meet the specs of the datasheet.
So that is a possible cause that cannot be excluded.

#### Thinking out loud

With your Riden RD6006P precise power supply, what is the smallest step you can set?

The INA226 should be able to see 36V in 65536 steps ==> steps of about 0.5 mV 
So if you can see that generate and see that difference one could test if the internal ADC is indeed 16 bit.

RobTillaart, 2023-12-04:
FYI, there are quite a bit of tech docs - https://www.ti.com/product/INA226#tech-docs
For the library I only used the datasheet, but the others might have some interesting data.
I will add this link in the readme.md

RobTillaart, 2023-12-04:
Have you tried **setBusVoltageConversionTime(7)** ?

RobTillaart, 2023-12-04:
Thanks for al these measurements, really appreciated. 
Although it does not give clues about the why of the deviation, they show detailed behavior on accuracy / precision.

zalexzperez, 2023-12-05:
Hello, I have news.
So far I had tried your library and the one by Peter Buchegger with similar results.

I found a blog entry [here](https://wolles-elektronikkiste.de/en/ina226-current-and-power-sensor) by the author of another INA226 library. Upon testing, I confirm it's giving accurate bus voltage readings without the error (35.00V when measuring 35V) and with at least two decimal positions of precision. 

Could you check it out and try to see what the difference is? Thanks

zalexzperez, 2023-12-05:
I deleted my previous message as I was announcing good news because I had tried a third library and thought it was finally giving the correct bus voltage. But it was just an old sketch running with a correction factor applied, sorry about that!

RobTillaart, 2023-12-05:
Maybe the 1.5% deviation is a accuracy problem and not a precision one.

Suppose I will implement corrections, how many would the library need?
- one for voltage
- one for current
- more?

These have typical the form of y = ax + b for linear correction. For power two linear corrections, I times V, will generate a quadratic one.

RobTillaart, 2023-12-05:
Pasted your measurements of the first post in spreadsheet to determine the correction formula.
This correction formula came out

```cpp
f(x) = 0.98529147 * x - 0.00633941587;
```

The maximum error is at 0 == 0.0231
Not counting in zero, the max error == 0.00625

As the b offset is rather small, (with some trial an error) this formula is almost equally good.
```cpp
f(x) = 0,98495 * x;
```

drawback of both formulas is that they "fail around zero"

zalexzperez, 2023-12-08:
Well,

I asked Wolfgang Ewald (owner of another INA226 library) about the issue and he replied his modules don't have such error. He also suggested I may have a fake one. Nevertheless,  since I'll build a custom PCB, I'll definitely get the genuine sensor and we'll find out if the error's still there or not.

Anyway, using a correction factor of 0.98525, 16 averages with 8244us bus conversion time gives me a good-enough precision of 5mV lower than the actual voltage in the use range that I will use this for (car battery voltage reading). I still have to see how it behaves inside of a car, though.

__________
Since you asked about other questions, I hope you don't mind my asking here about something else.
I've been busy trying to understand the calibration register and how it can supposedly increase the current resolution.  But first, let me inform you that the shunt voltage reading doesn't have a voltage error like the bus voltage does.

The datasheet states: "The highest resolution for the Current Register (04h) can be obtained by using the smallest allowable Current_LSB based on the maximum expected current"

So I started using the formulas trying to see how a lower expected current could give a higher current resolution, but didn't see how that was possible: Ohm's law dictates that the minimum measurable current increment should only depend on the shunt voltage LSB and the shunt resistor value:  2.5uV/R_shunt. If the shunt resistor is 0.1Ohms, the minimum current increment will be 2.5E-6/0.1=25uA. 

Now, we'll see if a lower "expected current" gives a higher current resolution or not with an example with two "maximum expected current" values and a real load current of 200.025mA.

- If the max expected current is around 0.8A:

 Current_LSB is rounded to 25uA/bit (0.8/2^15), giving a CAL register = 2048. 
Since load current is 0.200025A, the shunt voltage will be 20.0025mV, so shunt voltage register = 8001, hence the current register will be 8001 too. Finally, the measured current will be 8001 x 25E-6 = 200.025mA

- If the max expected current is around 0.4A:

Current_LSB is rounded to 12.5uA/bit (0.4/2^15), giving a CAL register = 4096.
Current register will be (8001X4096)/2048 = 16002. The measured current will be 16002 x 12.5E-6= 200.025mA.

As seen, both max expected current values result in the same measured current value. So I'm wondering what is the purpose of current_LSB parameter.

RobTillaart, 2023-12-09:
- Which library does Wolfgang Ewald use? Can he confirm this one is working properly?
- When you write numbers like 200.025 please be aware of the limits of the IEEE754 float operations in terms of significant digits. In theory a float has a 23 bits mantissa which equals 6 significant digits at most. In practice after some calculations you have only 4 or 5 left, e.g. if used values are approximations (like from an ADC) or if the floats differ in order of magnitude and you do additions.

----

When making a measurement with any device you have a number of specifications
- range, accuracy, precision, resolution (latter two are very related)

The calibration value of the INA226 defines the step size of the measurements a.k.a. the current_lsb  (precision).
If the current_lsb is smaller, the possible range is also smaller. These go hand in hand.
And yes there are settings that overlap and work well for different uses.

Suppose you have a 20A sensor and you know that you will never have more than 6 A.
You can change the calibration value to get a smaller current_LSB and thus a higher precision.
However this can only be obtained by having more samples (averaging etc). 
Note that a lower current_lsb is also more sensitive to noise. 
So in practice you need to do your math and do experiments to maximize the quality of the measurements if you want to go beyond 4 significant digits. Details are in the datasheet.

RobTillaart, 2023-12-09:
Your example (0.8A and 0.4 A) misses the point as the real value is a multiple of the highest currentLSB

Use a real load current of 200.015mA.

Max A = 0.8
CAL register = 2048.
it will measure 8001 => 200.025mA.  (deviates 10 uA)

Max A = 0.4
CAL register = 4096.
it will measure 16001 => 200.012.5mA. (deviates 2.5 uA so more precise than above.

you could do the math for A = 250 mA and it might be even closer.

zalexzperez, 2023-12-09:
> * Which library does Wolfgang Ewald use? Can he confirm this one is working properly?

https://github.com/wollewald/INA226_WE
I contacted him through a blog post and he said his modules don't exhibit the error. 

> it will measure 16001 => 200.012.5mA. (deviates 2.5 uA so more precise than above.

But how exactly can the current register be 16001? 
Shunt voltage register is 8001 for both max expected currents, as this measurement is independent from the calibration.  
So when the max expected current is 0.4, the current register should be (8001 x 4096) / 2048 = 16002 ! And for each voltage register bit increase, the current register will increase by two, due to the 4096/2048 ratio...

RobTillaart, 2023-12-09:
From datasheet:

[image]

> So when the max expected current is 0.4, the current register should be (8001 x 4096) / 2048 = 16002

Your math is correct if there was only a single measurement, however the sensor can make multiple and average them.
Otherwise it would indeed not make sense.

I also don't know if the internal math is 16 bit or 32 bit as your calculation assumes. 
Only if  I had more time (and hardware) to investigate such details.

RobTillaart, 2023-12-23:
@zalexzperez 
Any progress to report?

zalexzperez, 2023-12-23:
Nothing else to report. 
I have to finish designing my PCB and it's going to take some time before I can test the sensor. 
I'll report my findings once that time comes.

zalexzperez, 2024-01-26:
@RobTillaart 

Hello, yes, and it's good news! 
I received my custom ESP32-S3 based PCBs and made some testing. Since I'm not using the standard pins for i2C with the S3, I used Wire.setPins() prior to calling Wire.begin();

The problem is gone. No more need to use a correction factor.
At 35V I'm measuring 35.006V, at 5V it's 5.000V, at 14.456V it's exactly 14.456V.

Here's the genuine INA226 chip from LCSC:
[image]

And here's the INA226 from the AliExpress module:
[image]

I even went and desoldered the fake chip and put a good one in its place, to be able to test the module with an older ESP32-E microcontroller and the results are good, just like with the S3 board.

closed 2024-01-26 by zalexzperez

RobTillaart, 2024-01-26:
Good to hear the issue is solved,

Thanks for reporting and the images!

pedroke, 2024-03-22:
Hi guys, I have to confirm the issue. I have 4 pieces from China, ordered each from different seller and on different date... Each piece reports different voltages, with pretty obvious error. I've been already asking on TI forum, but just here I found the final confirmation those are really fakes :( Thanks for your efforts... we should spread this info...

RobTillaart, 2024-03-22:
Thanks for your addition.