RobTillaart/INA226 #35: Add "bool INA226::isReady()"

HenkHoldijk, 2024-01-06:
Not really an issue (more a DR) but could you add the following function to the library :

bool INA226::isReady()
{
  return ((_readRegister(INA226_MASK_ENABLE) & INA226_CONVERSION_READY_FLAG) == INA226_CONVERSION_READY_FLAG);
}

RobTillaart, 2024-01-06:
Yes, should be no problem.
Might be next week before I have time.

RobTillaart, 2024-01-06:
Are you OK to call it **bool isConversionReady()**, slightly more descriptive?

RobTillaart, 2024-01-06:
Created a develop branch with the function added. 
Can you verify it, if OK I can make a PR of it to release this.

PS, I have no (simple) example how /when to use it.

HenkHoldijk, 2024-01-06:
I'm OK with any name of the function that makes sense.

I'm using it in an application where the ESP shuts-down the INA226, then the ESP goes into deep sleep mode. After a while, it wakes-up re-configures the INA226 again waits for the conversion to be ready (using isReady) does some quick measurements (also using an ADS115) and shuts-down the INA226 .... Part of my usage : 

  // Wait until conversion is ready (but not longer than INA226_MAX_WAIT_MS ms)
  time_t dMillisStartWait = millis();
  while (!ina226.isReady() && ((millis() - dMillisStartWait) <= INA226_MAX_WAIT_MS)) yield();
  if ((millis() - dMillisStartWait) > INA226_MAX_WAIT_MS)
  {
    Serial.println("INA226 : Not ready in time (GetPackVoltage)");
    return NAN;
  }

RobTillaart, 2024-01-06:
INA226_MAX_WAIT_MS is defined by you?

This could be a library function too, something like

```cpp
bool waitConversionReady(uint32_t timeout = INA226_MAX_WAIT_MS)
{
  uint32_t start = millis();
  while ( (millis() - start) <= timeout) 
  {
    if (isConversionReady()) return true;
   delay(1);  //  implicit yield();
  }
  return false;
}
```

you could only write
```cpp
if (ina226.waitConversionReady() == false)   // uses default INA226_MAX_WAIT_MS
{
  Serial.println("INA226 : Not ready in time (GetPackVoltage)");
}
```

HenkHoldijk, 2024-01-06:
Right now waiting for the compilation of the isConversionReady test.

Did a Winmerge:
1) Comment in CHANGELOG.md still has the function name isReady
2) The order of the functions in the cpp-file is different from the order in the h-file

I'll add the waitConversionReady() to my local lib and will test that too.

HenkHoldijk, 2024-01-06:
isConversionReady() Works as it is supposed to work.

HenkHoldijk, 2024-01-06:
h-file :
#define INA226_MAX_WAIT_MS              ( 600 )
bool     waitConversionReady(uint32_t timeout = INA226_MAX_WAIT_MS);

cpp-file :
bool INA226::waitConversionReady(uint32_t timeout)
{
  uint32_t start = millis();
  while ((millis() - start) <= timeout) 
  {
    if (isConversionReady()) return true;
    delay(1);
  }
  return false;
}

Works for both Test-to-Pass & Test-to-Fail, but it does not give me a happy feeling.
After all the H-file requires a definition of the INA226_DEFAULT_TIMEOUT (which is determined by the configuration)

I'm now adding the class members :
  int8_t   _mode;
  int8_t   _avg;
  int8_t   _bvct;
  int8_t   _svct;
They are signed on purpose so they can be set to -1 in the constructor, allowing to see if functions are called.
I'll have a go in determining the timeout by using the above class members, which will then result in 2 
waitConversionReady functions (one with a parameter and one parameterless).

===

Not to be picky, but I also do not understand why some definitions are defined in the include file. They seem to be 
not needed for the library API and are of no use to the user of the INA226 library. I personally would hide them in 
the cpp-file or create a separate include file for these (and others). Lately I get bitten too many times by my own 
code in such a way that I even consider putting the class definition in a separate file.

RobTillaart, 2024-01-06:
> Comment in CHANGELOG.md still has the function name isReady

Fixed.

> The order of the functions in the cpp-file is different from the order in the h-file

Minor one, do you have compiler problems?

> Remove the F() from the h-file (INA226_LIB_VERSION) it will result in errors in combination with String()

Fixed. (please note that the String class can make Swiss chees from your memory)

>  In the h-file be consistent in using brackets for defines

Fixed, removed all. 
I have a cleanup in my backlog to replace all #defines by const int unless it is a parameter for the library. 
That would allow compile time checking too  (est ~500 files).

> Update the dates in the cpp-file and h-file

I use those as the start date of the libraries, but frankly think they should be removed as the version implies the date.
It is also in my backlog (est ~1100 files)

> The INA226_MAX_WAIT_MS b.t.w. is defined by me.
It can actually be calculated based on the conversion times and the average setting.
Adding 3 additional class members will make that more easy.

Should this include the SMBus Timeout  too => max 35 ms ?

RobTillaart, 2024-01-06:
Yes, there is a lot to improve to bring the lib to pro level. And that is for most if not all of my Arduino libraries. 
Would take another year or more of my free time I guess to update them all (if no new issues come in)

HenkHoldijk, 2024-01-06:
Right now I'm adding several definitions for the conversion times & avg. 
I'll also add the 2 versions of the waitConversionReady.

Mmm, I'm postponed learning how to do pull-requests on GitHub (already had to learn too many CMS (Configuration Management Systems), but maybe ...

HenkHoldijk, 2024-01-07:
Changed quite a bit and decided to take a leap and create a pull-request. ;)

Let's see if you like the changes.

HenkHoldijk, 2024-01-08:
B.t.w. If you want some help converting the examples, let me know.

RobTillaart, 2024-01-08:
Thanks for the offer.

Coming days I have scheduled work on other libraries and projects, will be back later this week.
Keep in mind that breaking changes might disrupt projects of many (non pro) users of the libraries.

commit 889e0e15 (2024-01-13): Develop (#37)

- Thanks to Henk Holdijk for his improvements.
- fix #35, add **bool isConversionReady()**
- add **bool waitConversionReady(uint32_t timeout = INA226_MAX_WAIT_MS)**
- add constant **INA226_MAX_WAIT_MS**
- changed return type **bool setAlertRegister(uint16_t mask)**
- changed return type **bool setAlertLimit(uint16_t limit)**
- reorder functions in .cpp file to match .h
- remove not needed include from examples.
- update readme.md
closed 2024-01-13 by RobTillaart