RobTillaart/INA226 #39: Can you provide an example of using multiple ina266

caoxuan19961206, 2024-04-01:
Can you provide an example of using multiple ina266，thankyou

RobTillaart, 2024-04-01:
Thanks for the question. I will look later this week

caoxuan19961206, 2024-04-01:
Thank you for all your assistance

RobTillaart, 2024-04-02:
Try this (compiles, bit not tested with hardware)
You should connect both devices slightly different to get different address (see datasheet for details).

```cpp
//    FILE: INA226_multi_device.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo three INA226's
//     URL: https://github.com/RobTillaart/INA226

#include "INA226.h"

INA226 INA0(0x40);
INA226 INA1(0x41);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA226_LIB_VERSION: ");
  Serial.println(INA226_LIB_VERSION);

  Wire.begin();
  if (!INA0.begin() )
  {
    Serial.println("INA0 could not connect. Fix and Reboot");
  }
  if (!INA1.begin() )
  {
    Serial.println("INA1 could not connect. Fix and Reboot");
  }

  INA0.setMaxCurrentShunt(1, 0.002);
  INA1.setMaxCurrentShunt(1, 0.002);
}

void loop()
{
  Serial.println("\nBUS\tSHUNT\tCURRENT\tPOWER\t\tBUS\tSHUNT\tCURRENT\tPOWER");
  for (int i = 0; i < 20; i++)
  {
    Serial.print(INA0.getBusVoltage(), 3);
    Serial.print("\t");
    Serial.print(INA0.getShuntVoltage_mV(), 3);
    Serial.print("\t");
    Serial.print(INA0.getCurrent_mA(), 3);
    Serial.print("\t");
    Serial.print(INA0.getPower_mW(), 3);
    Serial.print("\t");
    Serial.print("\t");
    Serial.print(INA1.getBusVoltage(), 3);
    Serial.print("\t");
    Serial.print(INA1.getShuntVoltage_mV(), 3);
    Serial.print("\t");
    Serial.print(INA1.getCurrent_mA(), 3);
    Serial.print("\t");
    Serial.print(INA1.getPower_mW(), 3);
    Serial.println();
    delay(1000);
  }
}

//  -- END OF FILE --
```

RobTillaart, 2024-04-02:
@caoxuan19961206 
Try this array of sensor variation - compiles, not tested with hardware
You might need to adjust the number of sensors.

Might be interesting to put multiple IN226 in series and see if they return the same current.

```cpp
//
//    FILE: INA226_array.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo four INA226's, adjust to your sensor count
//     URL: https://github.com/RobTillaart/INA226

#include "INA226.h"

#define INA_COUNT  4

INA226 INA[INA_COUNT] =
{
  INA226(0x40),
  INA226(0x41),
  INA226(0x42),
  INA226(0x43)
};

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA226_LIB_VERSION: ");
  Serial.println(INA226_LIB_VERSION);

  Wire.begin();
  bool failed = false;
  for (int ID = 0; ID < INA_COUNT; ID++)
  {
    if (!INA[ID].begin() )
    {
      failed = true;
      Serial.println(ID);
    }
    INA[ID].setMaxCurrentShunt(1, 0.002);
  }
  if (failed)
  {
    Serial.println("One or more INA could not connect. Fix and Reboot");
    while (1);
  }
}

void loop()
{
  Serial.println("\nID\tBUS\tSHUNT\tCURRENT\tPOWER");
  for (int i = 0; i < 20; i++)
  {
    for (int ID = 0; ID < 4; ID++)
    {
      Serial.print(ID);
      Serial.print("\t");
      Serial.print(INA[ID].getBusVoltage(), 3);
      Serial.print("\t");
      Serial.print(INA[ID].getShuntVoltage_mV(), 3);
      Serial.print("\t");
      Serial.print(INA[ID].getCurrent_mA(), 3);
      Serial.print("\t");
      Serial.print(INA[ID].getPower_mW(), 3);
      Serial.println();
      delay(1000);
    }
    Serial.println();
  }
}

//  -- END OF FILE --
```

RobTillaart, 2024-04-02:
Added examples to develop branch for next (minor) release.

caoxuan19961206, 2024-04-03:
Thank you for your reply, it has been tested

RobTillaart, 2024-04-03:
Did you verify both examples?

caoxuan19961206, 2024-04-03:
Tested, ready to run`11:22:12:232 -> 0	0.475	35.212	6553.400	3115.000
11:22:13:234 -> 1	0.000	-0.005	-2.400	0.000
11:22:14:238 -> 2	0.000	-0.005	-2.400	0.000
11:22:15:241 -> 3	0.000	0.000	0.000	0.000
11:22:16:243 -> 4	0.001	-0.005	-2.400	0.000
11:22:17:247 -> 5	0.001	-0.007	-5.000	0.000
11:22:18:250 -> `

caoxuan19961206, 2024-04-03:
Tested, ready to run
[image]

RobTillaart, 2024-04-04:
examples merged in 0.5.4

closed 2024-04-04 by RobTillaart