RobTillaart/INA226 #44: How to set ina.setMaxCurrentShunt for R100 shunt?

Tech500, 2024-04-29:
Requesting clarification on parameters for ina.setMaxCurrentShunt() using R100 shunt?  Do not understand how to arrive at values; will be measuring 12 volts.  Load current ~ 180 mA.

Load is a buck converter; going from 12 volts to 5 volts, 5 volts is connected to a video camera drawing 180 mA.

William

RobTillaart, 2024-04-29:
No hardware nearby to verify

I would try **INA.setMaxCurrentShunt(1.0, float ohm = 0.100)**.  Note however the 180 mA is quite small compared to the working range of the INA226. (~1%). So fluctuations in the current are even smaller. Think I would prefer an INA3221 (max 1.6A) as then the 180 mA would be a ~10%.

Never used the INA to measure a buck convertor. These devices have a dynamic load as they are switching on/off constantly. On a scope this might be visible as noise. See e.g. https://www.researchgate.net/figure/Experimental-buck-converter-Fig-2-Typical-output-voltage-load-transient-Average_fig1_286813528

HenkHoldijk, 2024-04-29:
For optimal dynamic range, the voltage over the shunt should be 80mV (actually 81.92mV but the lib limits this to 80mV) for your maximum load of 180mA. In this case I would remove the R100 shunt on your module, and replace it by E330 (330mOhm). In that case the voltage over the shunt will be around 60mV (0.180A * 0.33Ohm). Why E330? Because it will provide some slack regarding the maximum current and it is an easy resistor value to acquire. B.t.w. you do not have to place it on the same spot as the SMD 1206. You can also use the 2 pins for the terminal block. Still, the shunt voltage should not pass the 80mV during normal operation. For a E330 shunt, this 80mV is reached at a current of about 242.424mA (0.08V / 0.33E).

So replace the shunt by E330 and use ina.setMaxCurrentShunt(0.24242, 0.33); and you should be fine. I would also look at the filtering the INA226 provides and determine the Standard Deviation of your measurements (1000+) to determine what the best settings are (at least that is what I did). There are also ways to add an analog filter to the current input, but likely that will not be needed.

RobTillaart, 2024-05-28:
As there is no active discussion I close the issue.

Please note that today the 0.6.0 version is released that has a fix on **setMaxCurrentShunt()**

closed 2024-05-28 by RobTillaart