RobTillaart/INA226 #46: I use this driver to my project ,and sample resistor is 5 mR, but the value read from INA226 is not right ?

justdomyself, 2024-05-22:
`
  Wire.setPins(5, 4);

  Wire.begin();
  if (!INA.begin() )
  {
    Serial.println("could not connect. Fix and Reboot");
  }

  int ret = INA.setMaxCurrentShunt(5, 5/1000.0, true);

  if(ret != INA226_ERR_NONE){
      Serial.printf("ret != INA226_ERR_NONE , ret = %d??????????????????????\r\n", ret);
   }
   else{
    Serial.printf("INA226_ERR_NONE, suc!!!!!!!!!!!!!!!!!!!!!!\r\n");
   }

   //INA.setAverage(10);

  dump_reg();
`

print message below:

calib 1400
INA226_ERR_NONE, suc!!!!!!!!!!!!!!!!!!!!!!

        REGISTER        VALUE   VALUE_X   
        CONFIGURATION:  13727   359F      
                SHUNT:  154     9A        
          BUS VOLTAGE:  10258   2810      
                POWER:  57      39
              CURRENT:  178     B2
          CALIBRATION:  5120    1400

config 0x0007, id 0x0006

BUS     SHUNT   CURRENT POWER
12.823  0.382   38.200  245.000

the power is  about  3~4 w, but the read read is only 245mv

RobTillaart, 2024-05-22:
Thanks for your question, 
As I'm quite busy at the moment It might take a few days before I can dive into it.

RobTillaart, 2024-05-26:
@justdomyself 
Still very busy, sorry. 

----

Can you try skip the normalization?

**int ret = INA.setMaxCurrentShunt(5, 5/1000.0, false);**

Do the measurements look better then? 
(hypothesis is that the normalization fails for given parameters).

-----

> BUS SHUNT CURRENT POWER
> 12.823 0.382 38.200 245.000

12.823 V  x  38.2 mA = 489 mW so only 10% of the 3-4W you expected.

----

Question: do you measure DC or AC?

RobTillaart, 2024-05-26:
went through the code with your settings **INA.setMaxCurrentShunt(5, 5/1000.0, true);**

(set the printdebug flag in .cpp  - line 206 )
```
normalize:	 true
initial current_LSB:	152.6 uA / bit
Pre-scale current_LSB:	152.6 uA / bit
After scale current_LSB:	200.0 uA / bit
Final current_LSB:	200.0 uA / bit
Calibration:	5120
Max current:	6.554 A
Shunt:	0.0050 Ohm
ShuntV:	0.0250 Volt
```

So the current_LSB = 200uA / bit  or 0.2 mA / bit , which is normalized to a nice round value.
and the calibration = 5120 (the same as you got)

BUS SHUNT CURRENT POWER
12.823 0.382 38.200 245.000
The shunt gets 0.382 mVolt over a resistor of 5 mOhm =>  I = U/R = 0.382 mV / 0.005 Ohm =>  76.4 mA

12.823V * 76.4 mA => 979.6 mW     (so nowhere near any value nor the 3-4W nor the 245mW)

If you set normalize to false, you should get 6711 as calibration value as it will work with the values you used to configure

```
normalize:	 false
initial current_LSB:	152.6 uA / bit
Final current_LSB:	152.6 uA / bit
Calibration:	6711
Max current:	5.000 A
Shunt:	0.0050 Ohm
ShuntV:	0.0250 Volt
```

----

Note that the getPower() function is called at a different moment in time as the voltage and or the current measurements.

justdomyself, 2024-05-27:
> If you do have 3.5 Watt of power and a BUS voltage is 12.823 V you should see 3500 mW / 12.823V == 272 mA (there was 38.2)
> 
> 272 mA and a shunt voltage of 0.382 mV ==> R = U/ I => 0.382 / 272 = 0.0014 ohm ?
> 
> Are you sure about the R value of 0.005 Ohm?
> 
> I cannot make the numbers "add up"

thanks  for your  anser.

I have   resolve the problem,  the ina226 is bad,  I change another chip ,  I works right now 。

but if want to  the number read from the chip is more stable , the  average times must set bigger ,and  sample time set larger too .

RobTillaart, 2024-05-27:
Ok, bad chip explains all
There are functions to adjust the measurements.
So issue solved, I will close the issue.

closed 2024-05-27 by RobTillaart