RobTillaart/INA226 #53: Loss of Decimal Places in getBusVoltage()

Achim1944, 2025-02-12:
When I used the INA226 library, I found that I got
- 12.00 Volt for a 3S LiPo of 12.14 Volt
- 7.00 Volt for a 2S LiPo of 7.98 Volt

The reason in my opinion is that in getBusVoltage() the computation is done with val, which is uint16_t, but at the end returned as  float, so the digits after the decimal point are lost.

Achim

RobTillaart, 2025-02-12:
Sounds like a bug so I labeled it as such.
Will look into it asap

Which release do you use?

RobTillaart, 2025-02-12:
Quick look, yes you got a real bug,
I will work on a PR right away!

Thanks for pointing out.
Fix will be there in 15 minutes/

RobTillaart, 2025-02-12:
new code 
```cpp
float INA226::getBusVoltage()
{
  uint16_t val = _readRegister(INA226_BUS_VOLTAGE);
  //    return val * 1.25e-3 * _bus_V_scaling_e4 / 10000;  //  fixed 1.25 mV
  float voltage = val * 1.25e-3;
  if (_bus_V_scaling_e4 != 10000)
  {
    voltage *= _bus_V_scaling_e4 * 1.0e-4;
  }
  return voltage;
}
```

RobTillaart, 2025-02-12:
@Achim1944 
Created develop branch and PR, build has started.

RobTillaart, 2025-02-12:
@Achim1944 
Could you verify the develop branch if it solves the problem?

Thanks

HenkHoldijk, 2025-02-12:
As far as I know the previous version of the function was:

float INA226::getBusVoltage()
{
  uint16_t val = _readRegister(INA226_BUS_VOLTAGE);
  return val * 1.25e-3;  //  fixed 1.25 mV
}

That works perfectly. The computation uses uint16_t val, but the result is put in a float (the return type).

RobTillaart, 2025-02-12:
I introduced the error myself.

There was a PR that allowed to add a correction factor in small steps.
After merging that PR I optimized the new code to reduce math operations, and released it too fast.

Achim1944, 2025-02-12:
I tested the develop branch. Now it is working nicely, thanks!

closed 2025-02-12 by Achim1944

RobTillaart, 2025-02-12:
@Achim1944 

>  I tested the develop branch. Now it is working nicely, thanks!

Thanks for testing and reporting the bug, appreciated!

commit 1bd66ada (2025-02-12): fix #53, bug in 0.6.2 getBusVoltage (#54)

- fix #53, BUG in 0.6.2 getBusVoltage() looses decimals.