RobTillaart/INA226 #57: 在void setup()内初始化地址

Warm-rain, 2025-03-10:
如何在void setup()内初始化地址，我想自定义地址，从储存中读取地址，然而储存是在void setup()内初始化的，而ina226地址是硬编码的

HenkHoldijk, 2025-03-10:
In the constructor, see INA226.cpp : INA226::INA226(const uint8_t address, TwoWire *wire)
See also uint8_t INA226::getAddress() for retreving the address.

Warm-rain, 2025-03-10:
但是这个方法不是全局可用，这个问题出在变量作用域上。如果我在setup()函数内部声明了INA0和INA1，它们只在setup()函数内有效，其他函数（如ReadSensorsTask）无法访问这些变量。
`void setup() {
  Wire.begin();
  Wire.setClock(100000);
  INA226 INA0(INA226address1, &Wire);
  INA226 INA1(INA226address2, &Wire);
}`

Warm-rain, 2025-03-10:
> In the constructor, see INA226.cpp : INA226::INA226(const uint8_t address, TwoWire *wire) See also uint8_t INA226::getAddress() for retreving the address.

Sir, can you please help me write an example when you have time? Thank you.

RobTillaart, 2025-03-10:
Thanks for the question,

Did a quick test without hardware and it should be possible at least this way.
Please confirm if it works (or not).

I will check if I can do it with a pointer, might safe some RAM.

```cpp
//
//    FILE: INA226_runtime.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo set address runtime in setup()
//     URL: https://github.com/RobTillaart/INA226
//
//  not tested with HW yet

#include "INA226.h"

INA226 INA(0x40);  //  create a global placeholder

void setup()
{
  while (!Serial);
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA226_LIB_VERSION: ");
  Serial.println(INA226_LIB_VERSION);
  Serial.println();

  INA = INA226(0x41);
  Serial.println(INA.getAddress());

  Wire.begin();
  INA.begin();

  INA.setMaxCurrentShunt(1, 0.002);
}

void loop()
{
  Serial.println("\nBUS\tSHUNT\tCURRENT\tPOWER");
  for (int i = 0; i < 20; i++)
  {
    Serial.print(INA.getBusVoltage(), 3);
    Serial.print("\t");
    Serial.print(INA.getShuntVoltage_mV(), 3);
    Serial.print("\t");
    Serial.print(INA.getCurrent_mA(), 3);
    Serial.print("\t");
    Serial.print(INA.getPower_mW(), 3);
    Serial.println();
    delay(1000);
  }
}

//  -- END OF FILE --
```

RobTillaart, 2025-03-10:
With a pointer to an INA226 object you could switch runtime between two global defined objects.

```cpp
//
//    FILE: INA226_pointer.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//     URL: https://github.com/RobTillaart/INA226
//
//  not tested with HW yet

#include "INA226.h"

INA226 X(0x40);
INA226 Y(0x41);

INA226 * INA;

void setup()
{
  while (!Serial);
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA226_LIB_VERSION: ");
  Serial.println(INA226_LIB_VERSION);
  Serial.println();

  INA = &X;
  Serial.println(INA->getAddress());
  INA = &Y;
  Serial.println(INA->getAddress());

  Wire.begin();
  INA->begin();    //  <<<<<<<<<<<<<<<  note the -> pointer notation.

  INA->setMaxCurrentShunt(1, 0.002);
}

void loop()
{
  Serial.println("\nBUS\tSHUNT\tCURRENT\tPOWER");
  for (int i = 0; i < 20; i++)
  {
    Serial.print(INA->getBusVoltage(), 3);
    Serial.print("\t");
    Serial.print(INA->getShuntVoltage_mV(), 3);
    Serial.print("\t");
    Serial.print(INA->getCurrent_mA(), 3);
    Serial.print("\t");
    Serial.print(INA->getPower_mW(), 3);
    Serial.println();
    delay(1000);
  }
}

//  -- END OF FILE --
```

Warm-rain, 2025-03-11:
Tested the code OK, able to reassign addresses within Setup.

Warm-rain, 2025-03-11:
If multiple addresses can be automatically scanned and allocated, it will be a revolutionary feature.

RobTillaart, 2025-03-11:
Why not just create an array of INA226 devices?
Then you can access them by index.

RobTillaart, 2025-03-13:
>  Tested the code OK, able to reassign addresses within Setup.

As the problem is solved I close the issue.