RobTillaart/INA226 #58: Alert Working?

ComanderKai77, 2025-03-15:
Hi there,

got somebody the example INA226_demo_alert.ino working?
`INA.getAlertFlag()` always returns 8 even if no power is connected to the chip.

RobTillaart, 2025-03-15:
Thanks for the question,
Will investigate asap, probably in a few days.

HenkHoldijk, 2025-03-15:
I have the alert working, but to be honest, I refactored the INA226 lib completely regarding the alert.

Some snippets:

////////////////////////////////////////////////////////
//
//  alert
//

bool INA226::setAlertPinPolarity(ina226_alert_pin_polarity_enum ePolarity)
{
  uint16_t dRegVal = _readRegister(INA226_MASK_ENABLE);
  dRegVal &= ~(INA226_ALERT_PIN_POLARITY_BIT);            // Keep the only the other bits
  dRegVal |= (ePolarity * INA226_ALERT_PIN_POLARITY_BIT); // Add the new bit (or not)
  if (_writeRegister(INA226_MASK_ENABLE, dRegVal)) return false;
  return true;
}

bool INA226::setAlertLatch(ina226_alert_latch_enum eLatch)
{
  uint16_t dRegVal = _readRegister(INA226_MASK_ENABLE);
  dRegVal &= ~(INA226_ALERT_LATCH_ENABLE_BIT);          // Keep the only the other bits
  dRegVal |= (eLatch * INA226_ALERT_LATCH_ENABLE_BIT);  // Add the new bit (or not)
  if (_writeRegister(INA226_MASK_ENABLE, dRegVal)) return false;
  return true;
}

bool INA226::getAlertFunctionFlag()
{
  uint16_t dRegVAL = _readRegister(INA226_MASK_ENABLE);
  return (dRegVAL & INA226_ALERT_FUNCTION_FLAG) == INA226_ALERT_FUNCTION_FLAG;
}

bool INA226::setAlertLimit(uint16_t limit)
{
  if (_writeRegister(INA226_ALERT_LIMIT, limit)) return false;
  return true;
}

uint16_t INA226::getAlertLimit()
{
  return _readRegister(INA226_ALERT_LIMIT);
}

bool INA226::setAlert(ina226_alert_enum eAlertType, float fAlertLimit)
{
  uint16_t dAlertLimit = 0;

  switch(eAlertType)
  {
    case INA226_SHUNT_OVER_VOLTAGE_MV  :
    case INA226_SHUNT_UNDER_VOLTAGE_MV :
      // +/- 81.92 mV = +/- 32768 
      if (fAlertLimit >= 81.92) break;
      dAlertLimit = fAlertLimit * (32768 / 81.92);
      break;
    case INA226_BUS_OVER_VOLTAGE_V  :
    case INA226_BUS_UNDER_VOLTAGE_V :
      // Fixed LSB = 1.25 mV with 40.96 as a maximum (0x7FFF = 32767)
      // Above contradicts, so assume the 1.25 mV is correct
      if (fAlertLimit >= 40.96) break;
      dAlertLimit = fAlertLimit * (32768 / 40.96);
      break;
    case INA226_POWER_OVER_LIMIT_W :
      // Power LSB = 25 x the [1 × 10–3 Current_LSB]
      dAlertLimit = fAlertLimit / (25 * _current_LSB);
      break;
    case INA226_CONVERSION_READY :
      // Avoid compiler warning
      break;
    case INA226_SHUNT_OVER_CURRENT_A  :
    case INA226_SHUNT_UNDER_CURRENT_A :
      // Current = Shunt Voltage / Shunt Resistor
      // LSb     = 81.92 mV / 32768    = 2.5 uV
      // LSb     = 2.5 uV / _shunt Ohm = ... uA
      // Limit   = 10^6 * ((A * _shunt) / 2.5)
      if (fAlertLimit > _maxCurrent) break;
      dAlertLimit = 1e6 * ((fAlertLimit * _shunt) / 2.5);
      break;
    case INA226_SHUNT_OVER_CURRENT_MA  :
    case INA226_SHUNT_UNDER_CURRENT_MA :
      // Limit = 10^6 * ((A * _shunt) / 2.5)
      // Limit = 10^3 * ((mA * _shunt) / 2.5)  
      if (fAlertLimit > (1000 * _maxCurrent)) break;
      dAlertLimit = 1e3 * ((fAlertLimit * _shunt) / 2.5);
      break;
  }

  // Write the required limit
  if (eAlertType != INA226_CONVERSION_READY)
  {
    if (_writeRegister(INA226_ALERT_LIMIT, dAlertLimit)) return false;
  }

  // Get the current register content
  uint16_t dRegVal  = _readRegister(INA226_MASK_ENABLE);
  dRegVal          &= ~(0xFC00);              // Delete upper 6 MSb and keep all other
  dRegVal          |= (0xFC00 & eAlertType);  // Add the new bits (only the 6 MSb)
  if (_writeRegister(INA226_MASK_ENABLE, dRegVal)) return false;

  return true;
}

RobTillaart, 2025-03-17:
@ComanderKai77 

> INA.getAlertFlag() always returns 8 

The value 8 == **INA226_CONVERSION_READY_FLAG**, which makes sense when measurements are made. From datasheet p26.

_Conversion Ready Flag
Bit 3 Although the device can be read at any time, and the data from the last conversion is available, the Conversion
Ready Flag bit is provided to help coordinate one-shot or triggered conversions. The Conversion Ready Flag bit is
set **after all conversions, averaging, and multiplications are complete**. Conversion Ready Flag bit clears under the
following conditions:
1.) Writing to the Configuration Register (except for Power-Down selection)
2.) Reading the Mask/Enable Register_

----

> INA.getAlertFlag() always returns 8 even if no power is connected to the chip.

What do you mean with **no power connected**?

- **no power connected** ==> there is no load connected to the INA, so no current.

In this case the conversions would still occur so the **INA226_CONVERSION_READY_FLAG** is correct (after every conversion).

- **no power connected** ==> there is no VCC + GND connected to the sensor.

In this case I would expect that the low level readRegister (I2C) would fail.
To test this case, can you add two lines to print the error variable after **getAlertFlag()** in the sketch?

```cpp
void loop()
{
  uint16_t flags = INA.getAlertFlag();
  Serial.print("error: \t");
  Serial.println(INA.getLastError());
  if (flags )
  {
    Serial.print("alarm:\t");
    Serial.println(INA.getShuntVoltage());
  }
}
```

please post 10~20 lines of output.

RobTillaart, 2025-03-17:
@HenkHoldijk 

Nice work to be able to set the flags individually.

A performance improvement could be to only write back to the register if needed.

```cpp
bool INA226::setAlertPinPolarity(ina226_alert_pin_polarity_enum ePolarity)
{
  uint16_t dRegVal = _readRegister(INA226_MASK_ENABLE);
  //  is bit correctly set?
  if ((dRegVal & INA226_ALERT_PIN_POLARITY_BIT) ==  (ePolarity * INA226_ALERT_PIN_POLARITY_BIT)) return true;

  //  otherwise flip the bit
  dRegVal ^= INA226_ALERT_PIN_POLARITY_BIT;
  if (_writeRegister(INA226_MASK_ENABLE, dRegVal)) return false;
  return true;
}
```

The **bool INA226::setAlert(ina226_alert_enum eAlertType, float fAlertLimit)** looks great, however its signature could be interpreted as that for every eAlertType a limit can be set.  (Yes, my Alert API is probably worse)

I have to reread the datasheet carefully to see what it means if multiple alertTypes are set SOL SUL BOL BUL etc as there is only one limit value to test against.

ComanderKai77, 2025-03-17:
That looks better.
Thank you.

When should the alarm flag go back to "normal"?

RobTillaart, 2025-03-17:
Think when you read the register AND the trigger situation is not longer there.

You need to check the data sheet for the details. Learning to read them is a great asset and no you dont have to understand all at once.

HenkHoldijk, 2025-03-17:
Thanks for liking the snippets. ;)

Next weekend I'll look at the changes that Rob made and revert them back into my fork.
Maybe some stuff can be made a bit faster, but the current compilers are quite smart.
Functions like setAlertPinPolarity will likely only be used once in the init phase and at that 
moment I prefer the readability.

I refactored other parts also quite a bit in order to get compiler errors instead of a crash in 
case wrong values are being used (that is why I like to use enum in the interfaces). The API's 
have changed but the difference is limited and can be fixed within minutes at application 
level. Also the waitConversionReady is working perfectly (it calculates the time-out limit, 
depending on the conversion settings).

closed 2025-03-18 by ComanderKai77