RobTillaart/INA226 #59: ESP32-S3 DevKit unable to connect INA226 using the latest library

7dy15, 2025-04-06:
When using the provided INA226 library demo,it always suggests could not connect,fix and reboot,the SDA is connected to the default I2C SDA pin 8 ,and the SCL pin is connected to the default I2C SCL pin 9,the address is 0x40.the Wire.begin() returns true,but the INA.begin() returns false.this demo works on my arduino uno board,but failed on the ESP32S3 Devkit,How can I handle it?thanks for your advice.

7dy15, 2025-04-06:
And also when I use your MCP4725 library ,the I2C connection is no problem and works normally with default I2C pin 8 and pin 9

RobTillaart, 2025-04-06:
Thanks for the issue, will investigate coming days (busy today)

Can you see the device with an I2C scanner?

7dy15, 2025-04-07:
> Thanks for the issue, will investigate coming days (busy today)
> 
> Can you see the device with an I2C scanner?

well,something interesting...I switch the power supply  of INA226 from 5V to 3.3V,and it can be detected at 0x44 and sends back datas,but even when I attach no load,the shunt voltage sent back is 81.92V,and the bus voltage jumps between 0 and 40.96V

RobTillaart, 2025-04-07:
@7dy15 

the INA226 should work from 2.7 to 5.5V according to the datasheet. So it should have worked on both voltages.

The first thing that comes in my mind is there might be a short circuit near the shunt (or other wires)
You can check with your phone camera / magnifier.

The reading of the shunt voltage is as simple as it can be
```
float INA226::getShuntVoltage()
{
  int16_t val = _readRegister(INA226_SHUNT_VOLTAGE);
  return val * 2.5e-6;   //  fixed 2.50 uV
}
```
so given you see 81.92 V the register should have had an value of 32.768.000 
that is way beyond what can be in a register / 16 bit int so there is something wrong in the reading.

A possible reason could be you are reading 81.92 milliVolt or microVolt, and not Volts.

Can you verify your code which call you make?

RobTillaart, 2025-04-07:
@7dy15 

In the title of the issue you mention **using the latest library**
Have you tried previous versions and if so which was the last version that did work?

7dy15, 2025-04-07:
Yes,I tried another INA226 module and the datas sent back were right,maybe I bought a defected one,and the voltage was indeed in milliVolt.Thanks for your help

RobTillaart, 2025-04-07:
You're welcome
As the issue is solved I close it now

closed 2025-04-07 by RobTillaart